react-native-camera/react-native-camera #3231: iOS: Weird focus behaviour

bkljajic, 2021-05-19:
# Question

**To Do First**
- [ ] Take a look in the [README](https://github.com/react-native-community/react-native-camera/blob/master/README.md)
- [ ] Take a look in the [docs](https://github.com/react-native-community/react-native-camera/blob/master/docs/RNCamera.md)
- [ ] Take a look in the [QA](https://github.com/react-native-community/react-native-camera/blob/master/docs/QA.md)

**Ask your Question**

I'm sorry if this might be an incorrect issue template, but here goes:

We've developed a focusing feature where the user can tap on screen to use `autoFocusPointOfInterest` to focus on a point.

But when focusing on that specific point, the focusing action presents itself in a pulsating manner. This would happen also beforehand when we only used `autoFocus` set by default. 

When compared to the iOS camera's focusing this presents itself as a much smoother action. I do realise that the native camera app probably uses proprietary APIs and approaches, but I was wondering if somebody could suggest where to look as far as the code on the native side goes, to perhaps influence this behaviour - *what to modify on the library's side*, as the client is adamant this is sub-par performance.

You can see the comparison here, and perhaps get a clearer picture of the pulsating happening:
### RN Camera implementation

https://user-images.githubusercontent.com/38661703/118788494-24fa8300-b894-11eb-8c59-c6a57fdbcb15.mp4

### iOS Camera 

https://user-images.githubusercontent.com/38661703/118788590-38a5e980-b894-11eb-99a2-3941de5b3934.mp4

**Tags**
focus, autofocus, taptofocus, autoFocusPointOfInterest

bkljajic, 2021-06-13:
Bump.
Anybody have a clue if we can influence this focus behaviour ?

bkljajic, 2021-08-04:
Updated the description to show new findings - this only happens when **FPS** is specified.

cristianoccazinsp, 2021-08-04:
Have you tried on a different device? It is definitely strange, but I don't see anything that would change the behaviour when fps is set in the linked PR.

I would dig into the linked PR's changes, and mess around with some of the config changes it does to nail it down to the exact issue.  I've seen some odd issues on older iphones with the camera (such as the camera quality breaking when using AR).

bkljajic, 2021-08-04:
Hi, thanks for the reply.
Yes, we had tried and replicated this on both an old iPhone 6s and a newer iPhone XS Max, and a few other iPhones.
I’ve read the changes the PR in question brings, but I’ve still not pinpointed where the issue is. 
It can either be due to the format selection or the dimensions. Will look further, I guess.
Any hint is appreciated,

bkljajic, 2021-08-04:
I seem to have resolved the issue by changing the following: 
```js
for (AVFrameRateRange *range in format.videoSupportedFrameRateRanges) {
    if (range.minFrameRate <= desiredFPS && desiredFPS <= range.maxFrameRate && format.autoFocusSystem == 2) {
        selectedFormat = format;
        maxWidth = formatWidth;
    }
}
```
The key is in the `format.autoFocusSystem == 2`. It searches for a format with a **phaseDetection** focus system, which looks more like what the iOS Camera App is using.
This is truly only a 'patch', and I think should be investigated further

cristianoccazinsp, 2021-08-04:
Do all devices support that? Might want to fallback to the same loop without the extra condition if nothing matches just in case a device does not support it.

bkljajic, 2021-08-04:
I've tested it on an old iPhone 6s, and a newer iPhone XS Max, both seem to be working.
We don't have much time to go into depths, there is much more the format selection can offer, such as ISO selection, etc, but this fix seems to work for the app in question for now.
It really merits more analysis, I agree.
Thanks for the help!

closed 2021-10-11 by stale[bot]

hamzaouhssain1993, 2021-12-22:
Same issue, many people are complaining about that so thanks for the patch

hamzaouhssain1993, 2021-12-23:
@bkljajic I noticed that this solution works only for new devices i guess, I tried with an iPod and it did not fix it,