react-native-camera/react-native-camera #3233: RNCamera onFacesDetected: some android devices not working

minhngo111196, 2021-05-21:
# Question
I am working on a feature that uses react-native-camera's onFacesDetected function, but on many Samsung devices it works fine and some others like Samsung Note 8, Xiaomi Redmi Pro 8, Vsmart Joy 3, .. . does not detect the user's face. onFacesDetected does not return any events.

Here 's my code
<RNCamera
          ref={ref => { this.camera = ref }}
          faceDetectionMode={RNCamera.Constants.FaceDetection.Mode.accurate}
          onFacesDetected={response => this.handleFaceDetected(response)} // not working on some android device ...
          onFaceDetectionError={response => this.handleFaceDetectedError(response)}
          faceDetectionClassifications={RNCamera.Constants.FaceDetection.Classifications.all}
          style={styles.preview}
          type={RNCamera.Constants.Type.front}
          captureAudio={false}
/>

yamila-fraiman, 2021-06-25:
Same issue any idea?

closed 2021-08-31 by stale[bot]

Standin-Alone, 2022-06-16:
onfacesdetected crashes my app. any solution?

NiraliRadadiya, 2022-10-10:
same issue here.....any idea?