react-native-camera/react-native-camera #3250: RN Camera crash on screen touch.

NischalShahi, 2021-06-09:
# Bug Report

**To Do First**

- [yes ] Did you try latest release?
- [yes ] Did you try master?
- [yes ] Did you look for existing matching issues?

**Platforms**

**Versions**

- Android: 10
- react-native-camera: v3.44.1
- react-native: 0.64.0
- react: 17.0.2
  

**Description/Current Behaviour**

When RNCamera is running (video/photo), if user touches on the screen (say if the screen goes dim and user clicks). The app crashes.

**Expected Behaviour**

**Steps to Reproduce**

Open RNCamera, touch the screen.

> Love react-native-camera? Please consider supporting our collective: 👉 https://opencollective.com/react-native-camera/donate
> Want this issue to be resolved faster? Please consider adding a bounty to it https://issuehunt.io/repos/33218414

NischalShahi, 2021-06-13:
> Any crash log?

06-13 09:45:10.401  2011  2040 W android.os.Debug: failed to get memory consumption info: -1
06-13 09:45:14.499  5095  5095 E InputEventReceiver: Exception dispatching input event.
06-13 09:45:14.499  5095  5095 E MessageQueue-JNI: Exception in MessageQueue callback: handleReceiveCallback
06-13 09:45:14.634  5095  5095 E MessageQueue-JNI: java.lang.IllegalArgumentException: Unable to find JSIModule for class UIManager
06-13 09:45:14.634  5095  5095 E MessageQueue-JNI: 	at com.facebook.react.bridge.JSIModuleRegistry.getModule(JSIModuleRegistry.java:24)
06-13 09:45:14.634  5095  5095 E MessageQueue-JNI: 	at com.facebook.react.bridge.CatalystInstanceImpl.getJSIModule(CatalystInstanceImpl.java:564)
06-13 09:45:14.634  5095  5095 E MessageQueue-JNI: 	at com.facebook.react.uimanager.UIManagerHelper.getUIManager(UIManagerHelper.java:90)
06-13 09:45:14.634  5095  5095 E MessageQueue-JNI: 	at com.facebook.react.uimanager.UIManagerHelper.getUIManager(UIManagerHelper.java:46)
06-13 09:45:14.634  5095  5095 E MessageQueue-JNI: 	at com.facebook.react.uimanager.UIManagerHelper.getUIManagerForReactTag(UIManagerHelper.java:40)
06-13 09:45:14.634  5095  5095 E MessageQueue-JNI: 	at com.facebook.react.animated.NativeAnimatedNodesManager.handleEvent(NativeAnimatedNodesManager.java:505)
06-13 09:45:14.634  5095  5095 E MessageQueue-JNI: 	at com.facebook.react.animated.NativeAnimatedNodesManager.onEventDispatch(NativeAnimatedNodesManager.java:483)
06-13 09:45:14.634  5095  5095 E MessageQueue-JNI: 	at com.facebook.react.uimanager.events.EventDispatcherImpl.dispatchEvent(EventDispatcherImpl.java:116)
06-13 09:45:14.634  5095  5095 E MessageQueue-JNI: 	at com.facebook.react.uimanager.JSTouchDispatcher.handleTouchEvent(JSTouchDispatcher.java:74)
06-13 09:45:14.634  5095  5095 E MessageQueue-JNI: 	at com.facebook.react.ReactRootView.dispatchJSTouchEvent(ReactRootView.java:283)
06-13 09:45:14.634  5095  5095 E MessageQueue-JNI: 	at com.facebook.react.ReactRootView.onInterceptTouchEvent(ReactRootView.java:201)
06-13 09:45:14.634  5095  5095 E MessageQueue-JNI: 	at android.view.ViewGroup.dispatchTouchEvent(ViewGroup.java:2609)
06-13 09:45:14.634  5095  5095 E MessageQueue-JNI: 	at android.view.ViewGroup.dispatchTransformedTouchEvent(ViewGroup.java:3060)
06-13 09:45:14.634  5095  5095 E MessageQueue-JNI: 	at android.view.ViewGroup.dispatchTouchEvent(ViewGroup.java:2698)
06-13 09:45:14.634  5095  5095 E MessageQueue-JNI: 	at android.view.ViewGroup.dispatchTransformedTouchEvent(ViewGroup.java:3060)
06-13 09:45:14.634  5095  5095 E MessageQueue-JNI: 	at android.view.ViewGroup.dispatchTouchEvent(ViewGroup.java:2698)
06-13 09:45:14.634  5095  5095 E MessageQueue-JNI: 	at android.view.ViewGroup.dispatchTransformedTouchEvent(ViewGroup.java:3060)
06-13 09:45:14.634  5095  5095 E MessageQueue-JNI: 	at android.view.ViewGroup.dispatchTouchEvent(ViewGroup.java:2698)
06-13 09:45:14.634  5095  5095 E MessageQueue-JNI: 	at android.view.ViewGroup.dispatchTransformedTouchEvent(ViewGroup.java:3060)
06-13 09:45:14.634  5095  5095 E MessageQueue-JNI: 	at android.view.ViewGroup.dispatchTouchEvent(ViewGroup.java:2698)
06-13 09:45:14.634  5095  5095 E MessageQueue-JNI: 	at android.view.ViewGroup.dispatchTransformedTouchEvent(ViewGroup.java:3060)
06-13 09:45:14.634  5095  5095 E MessageQueue-JNI: 	at android.view.ViewGroup.dispatchTouchEvent(ViewGroup.java:2698)
06-13 09:45:14.634  5095  5095 E MessageQueue-JNI: 	at com.android.internal.policy.DecorView.superDispatchTouchEvent(DecorView.java:465)
06-13 09:45:14.634  5095  5095 E MessageQueue-JNI: 	at com.android.internal.policy.PhoneWindow.superDispatchTouchEvent(PhoneWindow.java:1849)
06-13 09:45:14.634  5095  5095 E MessageQueue-JNI: 	at android.app.Activity.dispatchTouchEvent(Activity.java:3993)
06-13 09:45:14.634  5095  5095 E MessageQueue-JNI: 	at androidx.appcompat.view.WindowCallbackWrapper.dispatchTouchEvent(WindowCallbackWrapper.java:69)
06-13 09:45:14.634  5095  5095 E MessageQueue-JNI: 	at com.android.internal.policy.DecorView.dispatchTouchEvent(DecorView.java:423)
06-13 09:45:14.634  5095  5095 E MessageQueue-JNI: 	at android.view.View.dispatchPointerEvent(View.java:13674)
06-13 09:45:14.634  5095  5095 E MessageQueue-JNI: 	at android.view.ViewRootImpl$ViewPostImeInputStage.processPointerEvent(ViewRootImpl.java:5482)
06-13 09:45:14.634  5095  5095 E MessageQueue-JNI: 	at android.view.ViewRootImpl$ViewPostImeInputStage.onProcess(ViewRootImpl.java:5285)
06-13 09:45:14.634  5095  5095 E MessageQueue-JNI: 	at android.view.ViewRootImpl$InputStage.deliver(ViewRootImpl.java:4788)
06-13 09:45:14.634  5095  5095 E MessageQueue-JNI: 	at android.view.ViewRootImpl$InputStage.onDeliverToNext(ViewRootImpl.java:4841)
06-13 09:45:14.634  5095  5095 E MessageQueue-JNI: 	at android.view.ViewRootImpl$InputStage.forward(ViewRootImpl.java:4807)
06-13 09:45:14.634  5095  5095 E MessageQueue-JNI: 	at android.view.ViewRootImpl$AsyncInputStage.forward(ViewRootImpl.java:4947)
06-13 09:45:14.634  5095  5095 E MessageQueue-JNI: 	at android.view.ViewRootImpl$InputStage.apply(ViewRootImpl.java:4815)
06-13 09:45:14.634  5095  5095 E MessageQueue-JNI: 	at android.view.ViewRootImpl$AsyncInputStage.apply(ViewRootImpl.java:5004)
06-13 09:45:14.634  5095  5095 E MessageQueue-JNI: 	at android.view.ViewRootImpl$InputStage.deliver(ViewRootImpl.java:4788)
06-13 09:45:14.634  5095  5095 E MessageQueue-JNI: 	at android.view.ViewRootImpl$InputStage.onDeliverToNext(ViewRootImpl.java:4841)
[... cut ...]
06-13 09:45:14.634  5095  5095 E MessageQueue-JNI: 	at android.os.MessageQueue.nativePollOnce(Native Method)
06-13 09:45:14.634  5095  5095 E MessageQueue-JNI: 	at android.os.MessageQueue.next(MessageQueue.java:336)
06-13 09:45:14.634  5095  5095 E MessageQueue-JNI: 	at android.os.Looper.loop(Looper.java:174)
06-13 09:45:14.634  5095  5095 E MessageQueue-JNI: 	at android.app.ActivityThread.main(ActivityThread.java:7356)
06-13 09:45:14.634  5095  5095 E MessageQueue-JNI: 	at java.lang.reflect.Method.invoke(Na
06-13 09:45:14.637  5095  5095 D AndroidRuntime: Shutting down VM

lizihanglove, 2021-06-16:
I got this too

Supun00SL, 2021-06-21:
is this issue fixed?

cristianoccazinsp, 2021-07-02:
So downgrading react is the solution? I guess RN 0.64 introduced a breaking change ?

pahoa3107, 2021-07-02:
upgrade react-native 0.64.2 to fix it

NischalShahi, 2021-07-02:
React Native 0.6.2 has fixed it.

closed 2021-07-02 by NischalShahi