react-native-camera/react-native-camera #3254: Difference in camera preview and final image

vishalenrique, 2021-06-10:
# Bug Report

**To Do First**

- [Y] Did you try latest release?
- [Y] Did you try master?
- [Y] Did you look for existing matching issues?

**Platforms**

- Android

**Versions**

- Android:7,8,9,10
- iOS:
- react-native-camera: 3.44.1
- react-native: 0.61.5
- react: 16.9.0
  

**Description/Current Behaviour**
Camera preview and final image seems different. Extra space gets added to the final picture which wasn't visible in camera preview.Not sure if it has something to do with aspect ratio, because when I give width/height matching with aspect ratio then the final image is same as preview one but when I use flex:1 to take all the available space then they're different.

**Expected Behaviour**
Final image should be same as the one seen in camera preview

**Steps to Reproduce**

> <View style={[{ flex: 1 }]}>
        <RNCamera
          ref={(ref) => {
            this.camera = ref;
          }}
          style={[
            styles.flexOne,
            {
              borderWidth: 6,
              borderColor: 'red',
              // width: 400,
              // height: 300,
              // flex: 0,
            },
          ]}
          ratio={'4:3'}>
          <View style={[{ position: 'absolute', top: 20, left: 0, right: 20, alignItems: 'flex-end' }]}>
            <Icon
              name={'cancel'}
              type={'material'}
              size={30}
              color={Colors.WHITE}
              onPress={debounce(() => Navigation.pop(this.props.componentId), 400, { trailing: false, leading: true })}
              underlayColor={'transparent'}
            />
          </View>
        </RNCamera>
      </View>

You can use the above code to see that the view doesn't end with the screen, to show that I've added borderWidth, when used with flex(Image 1) but when you use with width/height(uncomment the width/height/flex code) it shows the complete view(image 2).I think the extra space which comes in the final image is the one missing from that borderWidth.So the question is how to use as much space as you want yet have no difference between preview and final image?

**Additionals**

[image]

[image]

cristianoccazinsp, 2021-06-11:
I'm guessing you are using android? Yes, you must match the camera aspect ratio as the final image will have that ratio.

On iOS, I believe the camera is adjusted to the actual view dimensions so that is not a problem.

LupulescuAlexandru, 2021-06-11:
How are we supposed to do that?

I tried using this solution: https://github.com/react-native-camera/react-native-camera/issues/1280 but it seems it just squishes the preview after setting the ratio. I have to specify though that I have set the styles to flex: 1.

```javascript
const DESIRED_RATIO = '16:9';
const prepareRatio = async () => {
            if (Platform.OS === 'android' && camera.current) {
                const ratios = await camera.current.getSupportedRatiosAsync();

                // See if the current device has your desired ratio, otherwise get the maximum supported one
                // Usually the last element of "ratios" is the maximum supported ratio
                const ratio = ratios.find((ratio) => ratio === DESIRED_RATIO) || ratios[ratios.length - 1];
                console.log(ratios);
                setRatio(ratio);
            }
};

<RNCamera
                ref={camera}
                captureAudio={false}
                autoFocus={RNCamera.Constants.AutoFocus.on}
                style={styles.preview}
                onCameraReady={prepareRatio}
                ratio={ratio}
                type={RNCamera.Constants.Type.back}
                flashMode={RNCamera.Constants.FlashMode.off}
                androidCameraPermissionOptions={{
                    title: 'Permission to use camera',
                    message: 'We need your permission to use your camera',
                    buttonPositive: 'Ok',
                    buttonNegative: 'Cancel',
                }}>

const styles = ScaledSheet.create({
    preview: {
        flex: 1,
    },
})
```

cristianoccazinsp, 2021-06-11:
I wish there was an easier way, but setting flex: 1 (or anything to cover the entire screen) won't give you the right ratio as screens are neither 16:9 nor 4:3. Which by the way, 4:3 should be your desired ratio for photos as 16:9 photos are extremely awkward.

Besides telling android which ratio to use, this is also what I use to set the view's dimensions:

```

this.state.aspectRatio = 4/3; //  or 16/9
// top/left centers the camera frame so it looks like the native camera
if (isPortrait) {
  let height = orientation.width * this.state.aspectRatio;
  cameraStyle = {
    position: 'absolute',
    top: Math.max(0, (mainViewStyle.height - height) / 2),
    left: 0,
    width: orientation.width,
    height: height,
  };
} else {
  let height = orientation.height;
  let width = height * this.state.aspectRatio;

  cameraStyle = {
    position: 'absolute',
    top: 0,
    left: Math.max(0, (mainViewStyle.width - width) / 2),
    width: width,
    height: height,
  };
}
```

If you find a nicer way, let me know. At some point I guess this math should be built in into a camera helper.

flogy, 2021-06-17:
@cristianoccazinsp So you just set the camera style and `ratio` and then it works for you? Nothing else? For me it will always take pictures with `4000x3000` resolution even if I set `ratio` to `'16:9'` and resize the camera accordingly. 🤔

cristianoccazinsp, 2021-06-17:
The camera style is for the preview, so it shows the live preview in the same ratio as the final capture will end up.  The preview also defines the ratio of the final photo for iOS. For android, however, the actual captured photo ratio will be the one given by the `ratio` prop, but if the device doesn't support 16:9, chances are it will end up with 4:3 regardless.

Have you tried other android devices? Can you confirm the device supports the ratio you are giving it?

flogy, 2021-06-17:
Thanks for your quick reply. So far I have only tested on a OnePlus 8 Pro with Android 11. The 16:9 format should be supported, at least it is in the returned array when fetching the supported ratios using this library.

cristianoccazinsp, 2021-06-17:
That is strange, perhaps the code that returns supported ratios is bugged or use a different camera API than the one used by RN (e.g., Camera1 vs Camera2), are you using Camera2 by any chance?

In my case, I'm not using the code to query the available ratios, I just set it and hope for the device to support it. On both a pixel 2 and Pixel 5 that worked fine as you can see in this capture:

[image]

You may need to dig into the Java code to see what's going on if it still doesn't work for you.

closed 2021-08-23 by stale[bot]