react-native-camera/react-native-camera #3263: Bug Report To Do First   Did you try latest release?  Did you try master?  Did you look for existing matching issues? Description/Current Behaviour

bachbonglong, 2021-06-23:
# Bug Report

**To Do First**

- [ ] Did you try latest release?
- [ ] Did you try master?
- [ ] Did you look for existing matching issues?

**Platforms** -iOS . 
- react-native-camera: git+https://git@github.com/react-native-community/react-native-camera.git
- react-native: 0.61.5
- react: react-navigation3

Every time I run the camera, it black screen . I tried closing the app and reopening it several times before it showed the Camera normally. In Android it work , but Iphone under iphone X and below is easy to happen black screen as above.

And the case when the Camera works normally on screen B, I back to screen A, then go to screen B, it goes black.

Has anyone encountered this problem and fixed it? I have tried reading another issue and following it to no avail.

adoniscoder, 2021-06-23:
I'm having the same issue

nielsmadan, 2021-06-23:
Also happening for me with:

```
    "react-native": "0.64.2",
    "@react-navigation/native": "^5.9.4",
```

On iOS, but only in release mode. 

On the screen where this is happening, I am first showing some other UI (a message saying you need to give permission and then a button where if the user clicks, I ask for the camera permission). Once the user gives permission, I show the camera, but I get a completely black screen (there is supposed to be a header above the camera, but that is also not shown). When I restart the app and go to the screen (with permissions already given) it seems to work.

This works completely fine in debug mode on device and simulator.

closed 2021-08-29 by stale[bot]