react-native-camera/react-native-camera #3265: > Task :react-native-camera:mergeMlkitReleaseResources FAILED

MohamedAhmed122, 2021-06-25:
# I am trying to build APK build and I am  facing this Error
after I run ` ./gradlew assembleRelease`
[image]

Any clue what's the problem ? , or maybe solution

leena-sharma, 2021-06-28:
@MohamedAhmed122  
hi, did you ever solve this error?
Facing same issue , while creating release build.

paschalidi, 2021-07-03:
+1 having the same problem at the moment. Any ideas?!

MohamedAhmed122, 2021-07-06:
@paschalid , @leena-sharma here is the [solution](https://stackoverflow.com/questions/68139592/error-building-apk-build-error-task-apppackagerelease-failed/68139752#68139752)

paschalidi, 2021-07-06:
True this worked for my project!

closed 2021-09-11 by stale[bot]