react-native-camera/react-native-camera #3275: Android Flash turns off after taking picture using FlashMode.torch

hamzaouhssain1993, 2021-07-12:
# Bug Report

**To Do First**

- [YES ] Did you try latest release?
- [YES ] Did you try master?
- [ YES] Did you look for existing matching issues?

**Platforms**

It happens only in some android Devices, for example in Redmi Note 5.

No issue in IOS noticed
**Versions**

- Android: Used Android 9, it might be also in others
- iOS:
- react-native-camera:
- react-native: 0.64.1
- react: 17.0.1
  

**Description/Current Behaviour**

It happens in some Android devices, after you open the camera and activate the flash with FlashMode.torch, it turns off right after taking a picture. 

**Expected Behaviour**

the expected is that the flash stays on all the time 

**Steps to Reproduce**

Take and android device, open the camera, activate the flash with the variable  FlashMode.torch, take a picture and the flash turns off.
 i do not know

theasc, 2021-07-12:
I also have the same Issue, happens on some android devices.

drash-course, 2021-07-12:
> I do not know
Same, same

drash-course, 2021-07-12:
> I do not know

Same, same...

cristianoccazinsp, 2021-07-16:
Are you using camera2? Which devices?

hamzaouhssain1993, 2021-07-22:
i guess react-native is using camera 2 and the device for example for us is Redmi Note 5.

cristianoccazinsp, 2021-07-22:
Try without using camera 2, it is too buggy.

eomttt, 2021-08-12:
@cristianoccazinsp how to try without using camera 2???