react-native-camera/react-native-camera #3280: Notitification by the OS: Camera is being used

AxayDixit, 2021-07-16:
Hello Devs,

I'm recording videos using this library and I've been seeing the notification saying "Camera is being used by your application"  even after the camera related work is done.

Suprisingly, I've noticed this only on a few OnePlus devices. 

Below is the implementation:

```
startRecording = async () => {
    this.setState({ recording: true });
    this.countRecordTime = setInterval(() => this.setState({ seconds: this.state.seconds + 1 }), 1000);
    this.remRecordTime = setInterval(() => this.setState({ remCounter: this.state.remCounter - 1 }), 1000);
    const cameraConfig = { maxDuration: this.state.maxDuration, mirrorVideo: false, quality: RNCamera.Constants.VideoQuality["480p"] };
    const data = await this.camera.recordAsync(cameraConfig);
    this.setState({ recording: false });
    this.setState({ remCounter: 90 });
    clearInterval(this.remRecordTime);
    CameraRoll.saveToCameraRoll(data.uri, 'video').then(onfulfilled => {
      console.log("URI:" + data.uri)
      console.log("VIDEO_DATA:" + JSON.stringify(data))
      clearInterval(this.activeInterval)
      KeepAwake.deactivate()
      if (this.state.flashMode)
        this.controlFlashMode()
      this.props.navigation.navigate('SomeScreen')
      // ToastAndroid.show(`New video path: ${onfulfilled}`, ToastAndroid.SHORT)
    }).catch(error => ToastAndroid.show(`${error.message}`, ToastAndroid.SHORT));
  }
```
```
stopRecording = () => {
    this.camera.stopRecording();
    clearInterval(this.countRecordTime);
    this.setState({ seconds: 0 });
    clearInterval(this.activeInterval);
    // clearInterval(this.remRecordTime);
    // this.setState({ remCounter: 3 });
  }
```

Just to give you a reference about stopRecording(), it gets called on when a flag related to recording i.e. recording flag used in startRecording() is set to false.

Can somebody please let me know if you have already faced this before or is there anything I can do from the code side to make it go away?

MateusAndrade, 2021-07-16:
Strange, which Android version is this occuring?

AxayDixit, 2021-07-19:
> Strange, which Android version is this happening?

Sorry for the late reply.

This is on android 11 and the devices are OnePlus 7 & 7T.