react-native-camera/react-native-camera #3291: recordAsync(): If `fps` is specified, format selection is unstable.

Sidebook, 2021-07-22:
# Bug Report

**To Do First**

- [x] Did you try latest release?
- [x] Did you try master?
- [x] Did you look for existing matching issues?

**Platforms**

iOS

**Versions**

- Android:
- iOS: 14.5.1
- react-native-camera: 4.0.0
- react-native: 0.64.2
- react: 17.0.2
  

**Description/Current Behaviour**

If `fps` is specified in the option of `recordAsync()`. Format selection looks unstable.
Specifically, [this condition](https://github.com/react-native-camera/react-native-camera/blob/215b5643ba7f8c5e291958a51f92be310589a2f1/ios/RN/RNCamera.m#L116
) checks that the width of the new format fits the current one, but not for height.
Also, seems it doesn't need to store `maxWidth` for me. Actually, with my device, a 1440 x 1920 px video has been recorded with `1080p` setting. It should record a 1080 x 1920 px video. I'm assuming this is from the above line is checking only the format width.

If it's intentional behavior, could I hear about the reason here a little bit?

Thank you!

**Expected Behaviour**
It should select a format that has the correct dimension.

**Steps to Reproduce**

Call `recordAsync({fps: 60}` with the 1080p `defaultVideoQuality` prop

**Additionals**

> Love react-native-camera? Please consider supporting our collective: 👉 https://opencollective.com/react-native-camera/donate
> Want this issue to be resolved faster? Please consider adding a bounty to it https://issuehunt.io/repos/33218414

Sidebook, 2021-07-22:
Created PR for this issue as well, just as an example fix I imagine.

closed 2021-08-09 by renatobenks

n1ru4l, 2021-08-17:
:tada: This issue has been resolved in version 4.0.2 :tada:

The release is available on:
- [npm package (@latest dist-tag)](https://www.npmjs.com/package/react-native-camera/v/4.0.2)
- [GitHub release](https://github.com/react-native-camera/react-native-camera/releases/tag/v4.0.2)

Your **[semantic-release](https://github.com/semantic-release/semantic-release)** bot :package::rocket:

PR #3292 (2021-08-09): fix the format finding logic when fps is specified
Fix #3291