react-native-camera/react-native-camera #3299: Android takePictureAsync() returns grey image

hamzaouhssain1993, 2021-07-30:
# Bug Report

**To Do First**

- [YES ] Did you try latest release?
- [YES ] Did you try master?
- [YES ] Did you look for existing matching issues?

**Platforms**

Taking picture in portrait with the tablet Samsung galaxy tab S2 running Android 7 gives back only a grey image

**Versions**

- Android: 7
- iOS: /
- react-native-camera: 3.44.3
- react-native:0.64.1
- react:17.0.1
  

**Description/Current Behaviour**

Everytime I take a portrait picture with Samsung galaxy tab S2 running Android 7 I get a grey image. 
I checked the logs  in android and I got this error:
D/skia: libjpeg error 117 <Corrupt JPEG data: premature end of data segment> from output_message

**Expected Behaviour**

**Steps to Reproduce**

Use Samsung galaxy tab S2 running Android 7 and try to take a picture

**Additionals**

[image]

> Love react-native-camera? Please consider supporting our collective: 👉 https://opencollective.com/react-native-camera/donate
> Want this issue to be resolved faster? Please consider adding a bounty to it https://issuehunt.io/repos/33218414

renatobenks, 2021-08-01:
duplicated by #3060

closed 2021-08-01 by renatobenks

hamzaouhssain1993, 2021-08-02:
I know it is a duplicate! but that one is close! so open it please, close the issues does not make them disappear

renatobenks, 2021-08-02:
The problem is exactly the same, therefore it's better to concern the entire discussion on only one issue, don't you agree? This issue will not disappear once it's mentioned in the original issue.

hamzaouhssain1993, 2021-08-03:
I understand your point, but now we have two CLOSED issues. My concern is to have at least one of them open, it can be this one or the older one, but one of them should be reopened.

renatobenks, 2021-08-03:
I just reopened the other one @hamzaouhssain1993, I didn't notice that was closed, thank you for letting me know. I'll give updates there!