react-native-camera/react-native-camera #3300: version 4.0.0 crashes on Android devices

Megatron4537, 2021-07-30:
# Bug Report

**To Do First**

- [x ] Did you try latest release?
- [x ] Did you try master?
- [x ] Did you look for existing matching issues?

**Platforms**
Android Only

**Versions**

- Android: ?
- react-native-camera: 4.0.0
- react-native: 64.2
- react: 17.0.1
- react-navigation: 5.0.0

**Description/Current Behaviour**
rendering <RNCamera> opens the camera but crashes after a second in development on Android devices only.

**Expected Behaviour**

should render and not crash

**Steps to Reproduce**
Render <RNCamera>

**Additionals**
Should I publish the output of `adb logcat` ?
It worked with 3.44.0

MateusAndrade, 2021-07-31:
Hi @Megatron4537, we already tagged this as breaking change exactly because of this migration it's now a [4.0.0](https://github.com/react-native-camera/react-native-camera/releases/tag/v4.0.0), but it looks like there hadn't been a proper change to the docs/we need to improve then.

Btw, did you manage to run the examples without any problem? Or the same issue persists?

jzsn2018, 2021-08-20:
I have meet same problem
    "react-native": "0.64.2",
    "react-native-camera": "^4.0.3",

Evotionce, 2021-08-24:
Hi @MateusAndrade,
I tried to run the [Example from RNCamera.md](https://github.com/react-native-camera/react-native-camera/blob/master/docs/RNCamera.md) and faced the same issue. Removing `onGoogleVisionBarcodesDetected` worked for me.
My Oneplus is running on oxygen 9.0.6.

isnotahippy, 2021-08-29:
Same issue on the latest version. Should we not use this prop aftr 4.0.0? Do the normal barcode scanner props use Google MLKit underneath now, not clear from the docs? Looking at logs + logcat from adb there I don't get any useful errors to help diagnose this unfortunately.

sundayhd, 2021-09-15:
Same problem here 😐
when I use onGoogleVisionBarcodesDetected, I see input from camera for 1 sec and then my app crash.
I don't see any errors on console..
Maybe someone have workaround for this issue?

kerbaypapilon, 2021-09-16:
I have version 3.43.1 but when I run TextDetector my app crashes too on Android devices. When I text detection set to true app crashes. Any solutions?

joaopauloarnold-duobox, 2021-09-23:
Try version 3.44.0, it worked for me

WrathChaos, 2021-10-05:
@MateusAndrade,
Can you please explain to us how to use `onGoogleVisionBarcodesDetected` on Android as well? It works great on iOS but we cannot use it on Android with the 4+ version.

MuhammadKaleem171, 2021-11-13:
> Same problem here 😐 when I use onGoogleVisionBarcodesDetected, I see input from camera for 1 sec and then my app crash. I don't see any errors on console.. Maybe someone have workaround for this issue?

@sundayhd  did you found solution ?

KthProg, 2021-11-27:
missingDimensionStrategy 'react-native-camera', 'mlkit' (instead of 'general') in the gradle file worked for me.

LaVielle, 2021-12-03:
adb logcat does output some relevant logs:
```
12-03 11:36:58.576 28101 28139 E AndroidRuntime: FATAL EXCEPTION: AsyncTask #1
12-03 11:36:58.576 28101 28139 E AndroidRuntime: Process: com.vytalmerchantapp, PID: 28101
12-03 11:36:58.576 28101 28139 E AndroidRuntime: java.lang.RuntimeException: An error occurred while executing doInBackground()
12-03 11:36:58.576 28101 28139 E AndroidRuntime: 	at android.os.AsyncTask$4.done(AsyncTask.java:415)
12-03 11:36:58.576 28101 28139 E AndroidRuntime: 	at java.util.concurrent.FutureTask.finishCompletion(FutureTask.java:383)
12-03 11:36:58.576 28101 28139 E AndroidRuntime: 	at java.util.concurrent.FutureTask.setException(FutureTask.java:252)
12-03 11:36:58.576 28101 28139 E AndroidRuntime: 	at java.util.concurrent.FutureTask.run(FutureTask.java:271)
12-03 11:36:58.576 28101 28139 E AndroidRuntime: 	at android.os.AsyncTask$SerialExecutor$1.run(AsyncTask.java:305)
12-03 11:36:58.576 28101 28139 E AndroidRuntime: 	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1167)
12-03 11:36:58.576 28101 28139 E AndroidRuntime: 	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:641)
12-03 11:36:58.576 28101 28139 E AndroidRuntime: 	at java.lang.Thread.run(Thread.java:923)
12-03 11:36:58.576 28101 28139 E AndroidRuntime: Caused by: java.lang.IllegalStateException: Task is not yet complete
12-03 11:36:58.576 28101 28139 E AndroidRuntime: 	at com.google.android.gms.common.internal.Preconditions.checkState(com.google.android.gms:play-services-basement@@17.6.0:2)
12-03 11:36:58.576 28101 28139 E AndroidRuntime: 	at com.google.android.gms.tasks.zzw.zzf(com.google.android.gms:play-services-tasks@@17.2.1:1)
12-03 11:36:58.576 28101 28139 E AndroidRuntime: 	at com.google.android.gms.tasks.zzw.getResult(com.google.android.gms:play-services-tasks@@17.2.1:1)
12-03 11:36:58.576 28101 28139 E AndroidRuntime: 	at org.reactnative.barcodedetector.RNBarcodeDetector.detect(RNBarcodeDetector.java:55)
12-03 11:36:58.576 28101 28139 E AndroidRuntime: 	at org.reactnative.camera.tasks.BarcodeDetectorAsyncTask.doInBackground(BarcodeDetectorAsyncTask.java:65)
12-03 11:36:58.576 28101 28139 E AndroidRuntime: 	at org.reactnative.camera.tasks.BarcodeDetectorAsyncTask.doInBackground(BarcodeDetectorAsyncTask.java:18)
12-03 11:36:58.576 28101 28139 E AndroidRuntime: 	at android.os.AsyncTask$3.call(AsyncTask.java:394)
12-03 11:36:58.576 28101 28139 E AndroidRuntime: 	at java.util.concurrent.FutureTask.run(FutureTask.java:266)
```

Going through the [install steps](https://react-native-camera.github.io/react-native-camera/docs/installation) and updating a few things, as mentioned in the answers above, solved the crash issue for me.

yadigarbz, 2023-06-11:
> `missingDimensionStrategy 'react-native-camera', 'mlkit'` (instead of 'general') in the gradle file worked for me.

Definitely works 👍