react-native-camera/react-native-camera #3304: With android API 30, Does this library support it?

IhorFedorkiv94, 2021-08-04:
# Question
In 2021, we are continuing with our annual target API level update, requiring new apps to target API level 30 (Android 11) in August and in November for all app updates.
[https://android-developers.googleblog.com/2020/11/new-android-app-bundle-and-target-api.html](url)

So some methods will be **not supported**. Do you know about this?

renatobenks, 2021-08-04:
I haven't noticed upcoming big changes that wouldn't be supported anymore so far, did you? Let us know your concerns, but for what I got, probably just a few further changes to support new versions of RN.

IhorFedorkiv94, 2021-08-05:
Sorry for my bad explaining) 
My question is - **_react-native-camera_** lib will work fine with Android SDK  30?
Currentlly  **_react-native-camera_** lib has 29 version.

[image]

renatobenks, 2021-08-05:
Well, I think so far I know, it will. However, this project is far from being able to make sure of anything as it has been changed a lot over the years, and was introduced so many features without any precedence, which it became very unstable. So, as standard, to make sure of it, we'll gonna have to make some tests, can you help us with?

IhorFedorkiv94, 2021-08-06:
Thanks for your answers. Currently, I can't help with this, but if I have some information I will let you know :)

renatobenks, 2021-08-09:
No problem! I'll try to arrange something and see how it looks likee

closed 2021-08-09 by renatobenks