react-native-camera/react-native-camera #3322: Error when taking picture on Android 12

mortend, 2021-08-23:
# Bug Report

**To Do First**

- [x] Did you try latest release?
- [x] Did you try master?
- [x] Did you look for existing matching issues?

**Platforms**

Android 12 (via emulator) - x86_64

**Versions**

4.0.3, 3.23.1

**Description/Current Behaviour**

* Unable to take picture via `takePictureAsync()` on Android 12 (via emulator) - getting the following error:

```
 LOG  [Error: Camera capture failed. Camera is already capturing.]
```

* Works fine when running on Android 11 (on physical device).

**Expected Behaviour**

* `takePictureAsync()` is expected to succeed.

**Steps to Reproduce**

* Reproduction case: https://github.com/mortend/rn-camera-test

cristianoccazinsp, 2021-08-30:
@mortend just tested this on an Android 12 (Pixel 5) device and taking pictures worked just fine. Note that I am not using camera2 API, and I'm using the `cameraId` prop to select a specific camera.

Video recording also worked fine, except when the user globally disables microphone access (a new feature). When the user has globally disabled the microphone, video recording stops right away due to the permission popup that appears.  Still debugging this as it may be just our own recording implementation.

mortend, 2021-08-31:
@cristianoccazinsp Woah, I am also able to take pictures in the Android 12 emulator after setting `cameraId` and choosing the other camera instead. Many thanks for that hint!

I am not familiar with camera2 API. Does it work on the Pixel 5 when omitting `cameraId` and just using the default camera?

It's a little bit unfortunate that apps don't work on Android 12 emulators when using default settings -- is the camera2 API causing the problem?

cristianoccazinsp, 2021-08-31:
I think the default settings use camera1 so that should not be the problem.

If it works setting the cameraId but it doesn’t when just using the default type (back or front), then there’s definitely an issue with android 12.

dagi3d, 2021-10-31:
Can confirm that saw the same issue with the emulator. With the physical device with Android 12 works fine