react-native-camera/react-native-camera #3338: Adding pod subspec for onGoogleVisionBarcodesDetected leads to invalid bundle under iOS

mschmid, 2021-09-03:
First of all: **thanks for the good work**, barcode scanning works really well using this library!

I have an issue, maybe someone can relate.
I added the subspec for barcode scanning to my iOS app yesterday, like stated in https://react-native-camera.github.io/react-native-camera/docs/installation.html:
```
pod 'react-native-camera', path: '../node_modules/react-native-camera', subspecs: [
  'BarcodeDetectorMLKit'
]
```
Since then, uploading to the app store gives me the following error:

> ITMS-90562: Invalid Bundle - The app cannot be processed because options not allowed to be embedded in bitcode are detected in the submission. It is likely that you are not building the app with the toolchain provided in Xcode. Rebuild your entire app with the latest GM Xcode and submit the app again.

Before adding the subspec the transmission gave no errors, I was able to use `onBarCodeRead`.

mschmid, 2021-09-03:
I was able to resolve this issue.
It can be traced back to GoogleMLKit because also a flutter lib using GoogleMLKit has the same issue.
I turned off Bitcode using `ENABLE_BITCODE = NO` in my app and use this as a workaround, until google hopefully fixes this issue.

closed 2021-09-03 by mschmid

bneigher, 2021-09-23:
yes this is indeed breaking builds in my app as well