react-native-camera/react-native-camera #3345: Crash When Navigate To Camera - java.lang.RuntimeException: An error occurred while executing doInBackground()

iyansr, 2021-09-15:
# Bug Report

**To Do First**

- [x] Did you try latest release?
- [x] Did you try master?
- [x] Did you look for existing matching issues?

**Platforms**

Android 

**Versions**

- Android: Android 10 (Emulator)
- iOS:
- react-native-camera: 4.2.0
- react-native: 0.64.1
- react: 17.0.1
- react-navigation: 5.9.4

**Description/Current Behaviour**

- Crash When navigate to Camera Screen

**Expected Behaviour**

- Working Fine

**Steps to Reproduce**

- Navigate to camera screen,
- The Camera is Appear for a Second then Crash

**Additionals**

code : 

```jsx
import React, { useEffect, useRef, useState } from 'react'
import { StyleSheet, Text, TouchableOpacity, View } from 'react-native'
import { RNCamera } from 'react-native-camera'
import { check, PERMISSIONS, RESULTS, request } from 'react-native-permissions'

const Camera = () => {
   const camera = useRef()
   const [status, setStatus] = useState(RESULTS.DENIED)

   const takePicture = async () => {
      if (camera.current) {
         const options = { quality: 0.5, base64: true }
         const data = await camera.current?.takePictureAsync(options)
         console.log(data.uri)
      }
   }

   useEffect(() => {
      check(PERMISSIONS.ANDROID.CAMERA)
         .then((result) => {
            switch (result) {
               case RESULTS.UNAVAILABLE:
                  console.log('This feature is not available (on this device / in this context)')
                  alert('This feature is not available (on this device / in this context)')
                  break
               case RESULTS.DENIED:
                  console.log('The permission has not been requested / is denied but requestable')
                  request(PERMISSIONS.ANDROID.CAMERA).then((permissionStatus) => {
                     setStatus(permissionStatus)
                  })
                  break
               case RESULTS.LIMITED:
                  console.log('The permission is limited: some actions are possible')
                  break
               case RESULTS.GRANTED:
                  console.log('The permission is granted')
                  break
               case RESULTS.BLOCKED:
                  console.log('The permission is denied and not requestable anymore')
                  break
            }
         })
         .catch((error) => {
            // …
         })
   }, [])

   // if (status === RESULTS.DENIED) return null

   return (
      <View style={styles.container}>
         <RNCamera
            ref={camera}
            style={styles.preview}
            type={RNCamera.Constants.Type.back}
            androidCameraPermissionOptions={{
               title: 'Permission to use camera',
               message: 'We need your permission to use your camera',
               buttonPositive: 'Ok',
               buttonNegative: 'Cancel',
            }}
            androidRecordAudioPermissionOptions={{
               title: 'Permission to use audio recording',
               message: 'We need your permission to use your audio',
               buttonPositive: 'Ok',
               buttonNegative: 'Cancel',
            }}
            onGoogleVisionBarcodesDetected={({ barcodes }) => {
               console.log(barcodes)
            }}
         />
         <View style={{ flex: 0, flexDirection: 'row', justifyContent: 'center' }}>
            <TouchableOpacity onPress={takePicture} style={styles.capture}>
               <Text style={{ fontSize: 14 }}> SNAP </Text>
            </TouchableOpacity>
         </View>
      </View>
   )
}

const styles = StyleSheet.create({
   container: {
      flex: 1,
      flexDirection: 'column',
      backgroundColor: 'black',
   },
   preview: {
      flex: 1,
      justifyContent: 'flex-end',
      alignItems: 'center',
   },
   capture: {
      flex: 0,
      backgroundColor: '#fff',
      borderRadius: 5,
      padding: 15,
      paddingHorizontal: 20,
      alignSelf: 'center',
      margin: 20,
   },
})

export default Camera

```

error : 
```
Name
FATAL EXCEPTION: AsyncTask #5
Reason
java.lang.RuntimeException: An error occurred while executing doInBackground()
Stacktrace
FATAL EXCEPTION: AsyncTask #5
Process: com.apnamd.health, PID: 8597
java.lang.RuntimeException: An error occurred while executing doInBackground()
at android.os.AsyncTask$4.done(AsyncTask.java:399)
at java.util.concurrent.FutureTask.finishCompletion(FutureTask.java:383)
at java.util.concurrent.FutureTask.setException(FutureTask.java:252)
at java.util.concurrent.FutureTask.run(FutureTask.java:271)
at android.os.AsyncTask$SerialExecutor$1.run(AsyncTask.java:289)
at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1167)
at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:641)
at java.lang.Thread.run(Thread.java:919)
Caused by: java.lang.IllegalStateException: Task is not yet complete
at com.google.android.gms.common.internal.Preconditions.checkState(com.google.android.gms:play-services-basement@@17.6.0:2)
at com.google.android.gms.tasks.zzw.zzf(com.google.android.gms:play-services-tasks@@17.2.1:1)
at com.google.android.gms.tasks.zzw.getResult(com.google.android.gms:play-services-tasks@@17.2.1:1)
at org.reactnative.barcodedetector.RNBarcodeDetector.detect(RNBarcodeDetector.java:55)
at org.reactnative.camera.tasks.BarcodeDetectorAsyncTask.doInBackground(BarcodeDetectorAsyncTask.java:65)
at org.reactnative.camera.tasks.BarcodeDetectorAsyncTask.doInBackground(BarcodeDetectorAsyncTask.java:18)
at android.os.AsyncTask$3.call(AsyncTask.java:378)
at java.util.concurrent.FutureTask.run(FutureTask.java:266)
... 4 more
```

> Love react-native-camera? Please consider supporting our collective: 👉 https://opencollective.com/react-native-camera/donate
> Want this issue to be resolved faster? Please consider adding a bounty to it https://issuehunt.io/repos/33218414

Anificient, 2021-09-15:
what are hte `missingDimensionStrategy` you have added in your defaultConfig of `build.gradle`?

eduardocarminati, 2021-09-15:
It happened to me after upgrading **react-native** to 0.65.1 and **react-native-camera** from 3.x to 4.2.0.

To fix it, I replaced this line:
`missingDimensionStrategy 'react-native-camera', 'general'`
with this line:
`missingDimensionStrategy 'react-native-camera', 'mlkit'`

iyansr, 2021-09-15:
> @iyansr what are the `missingDimensionStrategy` you have added in your defaultConfig of `build.gradle`?

Its `react-native-camera`, `general`

iyansr, 2021-09-15:
> It happened to me after upgrading **react-native** to 0.65.1 and **react-native-camera** from 3.x to 4.2.0.
> 
> To fix it, I replaced this line:
> `missingDimensionStrategy 'react-native-camera', 'general'`
> with this line:
> `missingDimensionStrategy 'react-native-camera', 'mlkit'`

I will try

closed 2021-09-15 by iyansr