react-native-camera/react-native-camera #3346: How to use the RNCamera on consecutive screens?

fedeerbes, 2021-09-19:
# Question
How to use the RNCamera on consecutive screens?

**To Do First**
- [x] Take a look in the [README](https://github.com/react-native-community/react-native-camera/blob/master/README.md)
- [x] Take a look in the [docs](https://github.com/react-native-community/react-native-camera/blob/master/docs/RNCamera.md)
- [x] Take a look in the [QA](https://github.com/react-native-community/react-native-camera/blob/master/docs/QA.md)

**Ask your Question**

We're implementing KYC module in my app and the user needs to take pictures of front ID, back ID and selfie.
Front ID screen => Back ID screen => Selfie
We have a screen for each step and we're unmonting RNCamera component when the user unfocus the screen, but we have issues almost constantly, when moving from one screen to another or when going back, a black screen appears and after a couple of seconds the app dies.

Device: Samsung SM-J600G
Android: 29
react-native-camera: 4.2.0
react-native: 0.64.0
@react-navigation/native: 5.9.8
@react-navigation/stack: 5.14.9
react-native-screens: 3.7.2

from logcat
```
2021-09-19 14:26:32.761 26022-26022/com.applemoncash D/ViewRootImpl@9b8f127[MainActivity]: ViewPostIme key 0
2021-09-19 14:26:32.839 26022-26022/com.applemoncash D/ViewRootImpl@9b8f127[MainActivity]: ViewPostIme key 1
2021-09-19 14:26:33.163 26022-26022/com.applemoncash D/Camera: app passed NULL surface
2021-09-19 14:26:33.168 26022-26022/com.applemoncash D/SensorManager: unregisterListener ::   
2021-09-19 14:26:33.212 26022-26022/com.applemoncash D/Camera: app passed NULL surface
2021-09-19 14:26:33.221 26022-26022/com.applemoncash E/CAMERA_1::: onSurfaceDestroyed preview cleanup failed
    java.io.IOException: setPreviewTexture failed
        at android.hardware.Camera.setPreviewSurface(Native Method)
        at android.hardware.Camera.setPreviewDisplay(Camera.java:718)
        at com.google.android.cameraview.Camera1$1.onSurfaceDestroyed(Camera1.java:182)
        at com.google.android.cameraview.PreviewImpl.dispatchSurfaceDestroyed(PreviewImpl.java:60)
        at com.google.android.cameraview.TextureViewPreview$1.onSurfaceTextureDestroyed(TextureViewPreview.java:59)
        at android.view.TextureView.releaseSurfaceTexture(TextureView.java:259)
        at android.view.TextureView.onDetachedFromWindowInternal(TextureView.java:230)
        at android.view.View.dispatchDetachedFromWindow(View.java:21363)
        at android.view.ViewGroup.dispatchDetachedFromWindow(ViewGroup.java:4641)
        at android.view.ViewGroup.dispatchDetachedFromWindow(ViewGroup.java:4641)
        at android.view.ViewGroup.dispatchDetachedFromWindow(ViewGroup.java:4641)
        at android.view.ViewGroup.dispatchDetachedFromWindow(ViewGroup.java:4641)
        at android.view.ViewGroup.dispatchDetachedFromWindow(ViewGroup.java:4641)
        at android.view.ViewGroup.dispatchDetachedFromWindow(ViewGroup.java:4641)
        at android.view.ViewGroup.dispatchDetachedFromWindow(ViewGroup.java:4641)
        at android.view.ViewGroup.dispatchDetachedFromWindow(ViewGroup.java:4641)
        at android.view.ViewGroup.removeViewInternal(ViewGroup.java:6287)
        at android.view.ViewGroup.removeViewAt(ViewGroup.java:6234)
        at androidx.fragment.app.FragmentManager.moveFragmentToExpectedState(FragmentManager.java:1446)
        at androidx.fragment.app.FragmentManager.moveToState(FragmentManager.java:1497)
        at androidx.fragment.app.FragmentManager.executeOpsTogether(FragmentManager.java:2007)
        at androidx.fragment.app.FragmentManager.removeRedundantOperationsAndExecute(FragmentManager.java:1953)
        at androidx.fragment.app.FragmentManager.execSingleAction(FragmentManager.java:1818)
        at androidx.fragment.app.BackStackRecord.commitNowAllowingStateLoss(BackStackRecord.java:303)
        at com.swmansion.rnscreens.ScreenContainer.moveToFront(ScreenContainer.kt:184)
        at com.swmansion.rnscreens.ScreenContainer.onUpdate(ScreenContainer.kt:333)
        at com.swmansion.rnscreens.ScreenContainer.performUpdates(ScreenContainer.kt:294)
        at com.swmansion.rnscreens.ScreenContainer.performUpdatesNow(ScreenContainer.kt:286)
        at com.swmansion.rnscreens.ScreenContainer.notifyChildUpdate(ScreenContainer.kt:85)
        at com.swmansion.rnscreens.Screen.setActivityState(Screen.kt:163)
        at com.swmansion.rnscreens.ScreenViewManager.setActivityState(ScreenViewManager.kt:39)
        at java.lang.reflect.Method.invoke(Native Method)
        at com.facebook.react.uimanager.ViewManagersPropertyCache$PropSetter.updateViewProp(ViewManagersPropertyCache.java:87)
        at com.facebook.react.uimanager.ViewManagerPropertyUpdater$FallbackViewManagerSetter.setProperty(ViewManagerPropertyUpdater.java:136)
        at com.facebook.react.uimanager.ViewManagerPropertyUpdater.updateProps(ViewManagerPropertyUpdater.java:56)
        at com.facebook.react.uimanager.ViewManager.updateProperties(ViewManager.java:49)
        at com.facebook.react.uimanager.NativeViewHierarchyManager.updateProperties(NativeViewHierarchyManager.java:143)
        at com.facebook.react.uimanager.UIImplementation.synchronouslyUpdateViewOnUIThread(UIImplementation.java:291)
        at com.facebook.react.uimanager.UIManagerModule.synchronouslyUpdateViewOnUIThread(UIManagerModule.java:417)
        at com.facebook.react.animated.PropsAnimatedNode.updateView(PropsAnimatedNode.java:114)
        at com.facebook.react.animated.NativeAnimatedNodesManager.updateNodes(NativeAnimatedNodesManager.java:655)
        at com.facebook.react.animated.NativeAnimatedNodesManager.runUpdates(NativeAnimatedNodesManager.java:562)
        at com.facebook.react.animated.NativeAnimatedModule$1.doFrameGuarded(NativeAnimatedModule.java:139)
        at com.facebook.react.uimanager.GuardedFrameCallback.doFrame(GuardedFrameCallback.java:29)
        at com.facebook.react.modules.core.ReactChoreographer$ReactChoreographerDispatcher.doFrame(ReactChoreographer.java:175)
2021-09-19 14:26:33.222 26022-26022/com.applemoncash E/CAMERA_1:::     at com.facebook.react.modules.core.ChoreographerCompat$FrameCallback$1.doFrame(ChoreographerCompat.java:85)
        at android.view.Choreographer$CallbackRecord.run(Choreographer.java:994)
        at android.view.Choreographer.doCallbacks(Choreographer.java:794)
        at android.view.Choreographer.doFrame(Choreographer.java:725)
        at android.view.Choreographer$FrameDisplayEventReceiver.run(Choreographer.java:981)
        at android.os.Handler.handleCallback(Handler.java:883)
        at android.os.Handler.dispatchMessage(Handler.java:100)
        at android.os.Looper.loop(Looper.java:237)
        at android.app.ActivityThread.main(ActivityThread.java:7948)
        at java.lang.reflect.Method.invoke(Native Method)
        at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:493)
        at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1075)
2021-09-19 14:26:33.223 26022-26022/com.applemoncash D/SensorManager: unregisterListener ::   
2021-09-19 14:26:33.233 26022-26022/com.applemoncash D/SensorManager: registerListener :: 0, LIS2DS Accelerometer, 200000, 0,  
2021-09-19 14:26:33.244 26022-26471/com.applemoncash E/BufferQueueProducer: [SurfaceTexture-0-26022-3] connect: BufferQueue has been abandoned
2021-09-19 14:26:33.244 26022-26471/com.applemoncash E/BufferQueueProducer: [SurfaceTexture-0-26022-3] setMaxDequeuedBufferCount: BufferQueue has been abandoned
2021-09-19 14:26:35.138 26022-26393/com.applemoncash W/CameraBase: An error occurred while connecting to camera 0: Status(-129, EX_TRANSACTION_FAILED): '-32: '
2021-09-19 14:26:35.139 26022-26046/com.applemoncash W/CameraBase: Camera service died!
2021-09-19 14:26:35.139 26022-26040/com.applemoncash W/CameraBase: mediaserver's remote binder Camera object died
2021-09-19 14:26:35.142 26022-26471/com.applemoncash E/Camera: Error 2
2021-09-19 14:26:35.142 26022-26039/com.applemoncash W/Camera-JNI: callback on dead camera object
2021-09-19 14:26:35.143 26022-26040/com.applemoncash W/Camera-JNI: callback on dead camera object
2021-09-19 14:26:35.349 26022-26022/com.applemoncash D/SensorManager: unregisterListener ::   
2021-09-19 14:26:37.300 26022-26022/com.applemoncash D/ViewRootImpl@9b8f127[MainActivity]: ViewPostIme pointer 0
2021-09-19 14:26:37.362 26022-26022/com.applemoncash D/ViewRootImpl@9b8f127[MainActivity]: ViewPostIme pointer 1
```

**Tags**

> Love react-native-camera? Please consider supporting our collective: 👉  https://opencollective.com/react-native-camera/donate
> Want this issue to be resolved faster? Please consider adding a bounty to it https://issuehunt.io/repos/33218414

MateusAndrade, 2021-09-21:
Check if #3331 fixes this. 😄

fedeerbes, 2021-09-21:
@MateusAndrade I tried this solution on one of my Samsung SM-J600G devices (where the problem is reproducible) and it works. I will close this. Thank you and @cristianoccazinsp

closed 2021-09-21 by fedeerbes