react-native-camera/react-native-camera #3348: I am having difficulty installing with other libs [ANDROID]

brunobrante, 2021-09-21:
I am using react-native-onesignal and react-native-maps, both use firebase, and it gives several incompatibility errors, when I use only the two libraries it works normally, only adding react-native-camera gives error. Can someone help me adjust my build.grandle ? 

`
buildscript {
    repositories {
        gradlePluginPortal()
    }
    dependencies {
        classpath 'gradle.plugin.com.onesignal:onesignal-gradle-plugin:[0.12.10, 0.99.99]'
    }
}

apply plugin: 'com.onesignal.androidsdk.onesignal-gradle-plugin'
apply plugin: "com.android.application"

import com.android.build.OutputFile

project.ext.react = [
    enableHermes: false,  // clean and rebuild if changing
]

apply from: "../../node_modules/react-native/react.gradle"

/**
 * Set this to true to create two separate APKs instead of one:
 *   - An APK that only works on ARM devices
 *   - An APK that only works on x86 devices
 * The advantage is the size of the APK is reduced by about 4MB.
 * Upload all the APKs to the Play Store and people will download
 * the correct one based on the CPU architecture of their device.
 */
def enableSeparateBuildPerCPUArchitecture = false

/**
 * Run Proguard to shrink the Java bytecode in release builds.
 */
def enableProguardInReleaseBuilds = false

/**
 * The preferred build flavor of JavaScriptCore.
 *
 * For example, to use the international variant, you can use:
 * `def jscFlavor = 'org.webkit:android-jsc-intl:+'`
 *
 * The international variant includes ICU i18n library and necessary data
 * allowing to use e.g. `Date.toLocaleString` and `String.localeCompare` that
 * give correct results when using with locales other than en-US.  Note that
 * this variant is about 6MiB larger per architecture than default.
 */
def jscFlavor = 'org.webkit:android-jsc:+'

/**
 * Whether to enable the Hermes VM.
 *
 * This should be set on project.ext.react and mirrored here.  If it is not set
 * on project.ext.react, JavaScript will not be compiled to Hermes Bytecode
 * and the benefits of using Hermes will therefore be sharply reduced.
 */
def enableHermes = project.ext.react.get("enableHermes", false);

android {
    ndkVersion rootProject.ext.ndkVersion

    compileSdkVersion rootProject.ext.compileSdkVersion

    defaultConfig {
        applicationId "com.teste"
        minSdkVersion rootProject.ext.minSdkVersion
        targetSdkVersion rootProject.ext.targetSdkVersion
        versionCode 1
        versionName "1.0"
		missingDimensionStrategy 'react-native-camera', 'mlkit'
    }
    splits {
        abi {
            reset()
            enable enableSeparateBuildPerCPUArchitecture
            universalApk false  // If true, also generate a universal APK
            include "armeabi-v7a", "x86", "arm64-v8a", "x86_64"
        }
    }
    signingConfigs {
        debug {
            storeFile file('debug.keystore')
            storePassword 'android'
            keyAlias 'androiddebugkey'
            keyPassword 'android'
        }
    }
    buildTypes {
        debug {
            signingConfig signingConfigs.debug
        }
        release {
            // Caution! In production, you need to generate your own keystore file.
            // see https://reactnative.dev/docs/signed-apk-android.
            signingConfig signingConfigs.debug
            minifyEnabled enableProguardInReleaseBuilds
            proguardFiles getDefaultProguardFile("proguard-android.txt"), "proguard-rules.pro"
        }
    }

    // applicationVariants are e.g. debug, release
    applicationVariants.all { variant ->
        variant.outputs.each { output ->
            // For each separate APK per architecture, set a unique version code as described here:
            // https://developer.android.com/studio/build/configure-apk-splits.html
            // Example: versionCode 1 will generate 1001 for armeabi-v7a, 1002 for x86, etc.
            def versionCodes = ["armeabi-v7a": 1, "x86": 2, "arm64-v8a": 3, "x86_64": 4]
            def abi = output.getFilter(OutputFile.ABI)
            if (abi != null) {  // null for the universal-debug, universal-release variants
                output.versionCodeOverride =
                        defaultConfig.versionCode * 1000 + versionCodes.get(abi)
            }

        }
    }
}

dependencies {
    implementation fileTree(dir: "libs", include: ["*.jar"])
    //noinspection GradleDynamicVersion
    implementation "com.facebook.react:react-native:+"  // From node_modules

    implementation "androidx.swiperefreshlayout:swiperefreshlayout:1.0.0"

    debugImplementation("com.facebook.flipper:flipper:${FLIPPER_VERSION}") {
      exclude group:'com.facebook.fbjni'
    }

    debugImplementation("com.facebook.flipper:flipper-network-plugin:${FLIPPER_VERSION}") {
        exclude group:'com.facebook.flipper'
        exclude group:'com.squareup.okhttp3', module:'okhttp'
    }

    debugImplementation("com.facebook.flipper:flipper-fresco-plugin:${FLIPPER_VERSION}") {
        exclude group:'com.facebook.flipper'
    }

	implementation(project(':react-native-onesignal')){
      exclude group: 'com.google.android.gms'
	}

	implementation(project(':react-native-maps')){
		exclude group: 'com.google.android.gms'
	}

	implementation 'com.google.android.gms:play-services-base:17.6.0'
	implementation 'com.google.android.gms:play-services-basement:17.6.0'
	implementation 'com.google.android.gms:play-services-location:18.0.0'
	implementation 'com.google.android.gms:play-services-tasks:17.2.1'
	implementation 'com.google.android.gms:play-services-maps:17.0.0'

    if (enableHermes) {
        def hermesPath = "../../node_modules/hermes-engine/android/";
        debugImplementation files(hermesPath + "hermes-debug.aar")
        releaseImplementation files(hermesPath + "hermes-release.aar")
    } else {
        implementation jscFlavor
    }
}

// Run this once to be able to run the application with BUCK
// puts all compile dependencies into folder libs for BUCK to use
task copyDownloadableDepsToLibs(type: Copy) {
    from configurations.implementation
    into 'libs'
}

apply from: file("../../node_modules/@react-native-community/cli-platform-android/native_modules.gradle"); applyNativeModulesAppBuildGradle(project)
`

closed 2021-09-22 by brunobrante

GuillermoVA, 2021-10-01:
Do you solve this problem?? I have exactly the same

brunobrante, 2021-10-04:
> Do you solve this problem?? I have exactly the same

Hello @GuillermoVA, yes I solved it, basically I had to give an exclude to the firabase group, and implemented the libs separately. 

it looked like this
```
	implementation(project(':react-native-maps')){
		exclude group: 'com.google.android.gms'
	}

	implementation(project(':react-native-onesignal')){
      	        exclude group: 'com.google.android.gms'
	  	exclude group: 'com.google.firebase' // <=== this
	}

	implementation(project(':react-native-camera')){
		exclude group: 'com.google.android.gms'
	}

	implementation 'com.google.android.gms:play-services-base:17.6.0'
	implementation 'com.google.android.gms:play-services-basement:17.6.0'
	implementation 'com.google.android.gms:play-services-location:18.0.0'
	implementation 'com.google.android.gms:play-services-tasks:17.2.1'
	implementation 'com.google.android.gms:play-services-maps:17.0.0'
	implementation 'com.google.firebase:firebase-messaging:22.0.0'  // <=== this
	implementation 'com.google.mlkit:barcode-scanning:17.0.0' // <=== this
```

brunobrante, 2021-10-04:
> > Do you solve this problem?? I have exactly the same
> 
> Hello @GuillermoVA, yes I solved it, basically I had to give an exclude to the firebase group, and implemented the libs separately.
> 
> it looked like this
> 
> ```
> 	implementation(project(':react-native-maps')){
> 		exclude group: 'com.google.android.gms'
> 	}
> 
> 	implementation(project(':react-native-onesignal')){
>       	        exclude group: 'com.google.android.gms'
> 	  	exclude group: 'com.google.firebase' // <=== this
> 	}
> 
> 	implementation(project(':react-native-camera')){
> 		exclude group: 'com.google.android.gms'
> 	}
> 
> 	implementation 'com.google.android.gms:play-services-base:17.6.0'
> 	implementation 'com.google.android.gms:play-services-basement:17.6.0'
> 	implementation 'com.google.android.gms:play-services-location:18.0.0'
> 	implementation 'com.google.android.gms:play-services-tasks:17.2.1'
> 	implementation 'com.google.android.gms:play-services-maps:17.0.0'
> 	implementation 'com.google.firebase:firebase-messaging:22.0.0'  // <=== this
> 	implementation 'com.google.mlkit:barcode-scanning:17.0.0' // <=== this
> ```

If you use other camera functionality, maybe you have to implement some more libs.

mrigo, 2021-11-21:
I can confirm this solves the problem!
Thanks @Pyroboomka you saved my life!
```

implementation(project(':react-native-camera')){
     exclude group: 'com.google.android.gms'
}
```