react-native-camera/react-native-camera #3365: Definition of 'MLKText' must be imported from module 'MLKitTextRecognitionCommon.MLKText' before it is required

tungduonghgg123, 2021-10-07:
# Bug Report

**To Do First**

- [ yes] Did you try latest release?
- [ yes] Did you try master?
- [ yes] Did you look for existing matching issues?

**Platforms**  
iOS

**Versions**

- iOS: version 15, xcode 13
- react-native-camera: ^4.0.0
- react-native: 0.63.4
- react: 16.13.1

**Description/Current Behaviour**

When building using xcode, got this error 
**Expected Behaviour**
[image]

**Steps to Reproduce**

**Additionals**
I can just add `@import MLKitTextRecognitionCommon;` to `TextDetectorManager` however, the library should be able to build without manually changing like this. Thanks

> Love react-native-camera? Please consider supporting our collective: 👉 https://opencollective.com/react-native-camera/donate
> Want this issue to be resolved faster? Please consider adding a bounty to it https://issuehunt.io/repos/33218414

mahamaad, 2021-11-18:
i fix this issue by import all ML packages

@import MLKitVision;
//add all imports
@import MLKitTextRecognition;
@import MLKitTextRecognitionCommon;
@import MLCompute;
@import MLKitVision;

Atlan8, 2021-12-17:
> 

working!!!!!

liuxh-fullstack, 2022-01-21:
> i fix this issue by import all ML packages
> 
> @import MLKitVision; //add all imports @import MLKitTextRecognition; @import MLKitTextRecognitionCommon; @import MLCompute; @import MLKitVision;

I can work for me ,thank you very much

Dassine, 2022-01-28:
Hi there! @liuxh-fullstack @mahamaad  where did you add these imports? I tried in `AppDelegate.h`  but it doesn't seem to work :/

NathanBilleau, 2022-02-06:
> i fix this issue by import all ML packages
> 
> @import MLKitVision; //add all imports @import MLKitTextRecognition; @import MLKitTextRecognitionCommon; @import MLCompute; @import MLKitVision;

It worked for a few time but I now have the error "module 'MLCompute' not found" (building for iPhone 12 simulator on M1 MacBook), any idea how to solve it ? 
[image]

eduardobonetti, 2022-04-20:
> It worked for a few time but I now have the error "module 'MLCompute' not found" (building for iPhone 12 simulator on M1 MacBook), any idea how to solve it ? [image]

I have the same issue, any solution?

mahamaad, 2022-04-20:
react-native-camera is deprecated instead of this you can use [react-native-vision-camera](https://github.com/mrousavy/react-native-vision-camera)