react-native-camera/react-native-camera #3370: Face Detection as well as record video at the same time

Invictus-Munish, 2021-10-14:
# Bug Report

**To Do First**

- [ ] Did you try latest release?
- [ ] Did you try master?
- [ ] Did you look for existing matching issues?

**Platforms**

**Versions**

- Android: 11
- iOS:
- react-native-camera: 4.2.1
- react-native: 0.65.1
- react: 17.0.2
  

**Description/Current Behaviour**

I cannot record video and do face detection concurrently. As soon as i start video recording face detection stops

**Expected Behaviour**

Both should work together

**Additionals**

> Love react-native-camera? Please consider supporting our collective: 👉 https://opencollective.com/react-native-camera/donate
> Want this issue to be resolved faster? Please consider adding a bounty to it https://issuehunt.io/repos/33218414

toannd2409, 2021-11-11:
same issue :(

Invictus-Munish, 2021-11-12:
> same issue :(

no solutions for it, u can work simultaneously but not at same time

Thushar-Sreenivas, 2021-12-14:
Same issue here. Can anyone here suggest workarounds or hacks to get around this issue?