react-native-camera/react-native-camera #3392: IOS CRASH  while camera set  with onGoogleVisionBarcodesDetected

blazerroadg, 2021-12-02:
# Bug Report

**To Do First**

- [ ] Did you try latest release? YES
- [ ] Did you try master? YES
- [ ] Did you look for existing matching issues? YES

**Platforms**
iOS

- iOS: 15
- react-native-camera:4.2.1
- react-native: 0.66.3
- react:17.0.2
react-navigation:6.0.6

**Description/Current Behaviour**
On first page render RNCAMERA with  onGoogleVisionBarcodesDetected  property sets, Navigate to new page which render camera also with onGoogleVisionBarcodesDetected set, when start recordAsync app will crash

**Expected Behaviour**
Camera start recording 

**Steps to Reproduce**
Create two page 
at first page render camera with onGoogleVisionBarcodesDetected navigate to second page  which render the same camera with onGoogleVisionBarcodesDetected set, start recording 

**Additionals**
Also if you try to switch camera type the camera will be freeze 

I SOLVED this issue by render second camera without onGoogleVisionBarcodesDetected set.

2814793482, 2022-01-19:
the same issue, but I must use onGoogleVisionBarcodesDetected because I want to get the image

jgalianoz, 2022-04-14:
Any update about this?

CodingWithNobody, 2023-04-15:
Try this https://www.youtube.com/watch?v=sE0MEyFA7E4&list=PLQhQEGkwKZUrempLnmxjt7ZCZJu1W3p2i&index=8