react-native-camera/react-native-camera #3393: Ios camera get stuck after first foto and can't resume

Ivan-Kayn, 2021-12-03:
# Bug Report

**To Do First**

- [ ] Did you try latest release? Yes
- [ ] Did you try master? Yes
- [ ] Did you look for existing matching issues? Yes

**Platforms**

**Versions**

- iOS: 15.0.2
- react-native-camera: "4.2.1"
- react-native: "0.63.4"
- react: "16.13.1"
  
  wix "react-native-navigation": "^7.8.4"

**Description/Current Behaviour**
Camera works fine first time but gets stuck after I take a photo and come back to make another one. Camera view become obscured and slow. If I try to take new photo takePictureAsync throws: [Error: Camera is not ready.].  Btw, camera status continue to be 'ready' and onMountError don't give any errors. The strange thing is that on my device is happening every time I open camera after app is loaded but on other ios devices it's happening rarely. 
Another strange thing is that If I try to make video when I open it for the first time video stops after a second but if I retry it's working fine and don't get stuck anymore.

**Expected Behaviour**
To work fine after first use.

**Steps to Reproduce**
Take one photo, then try to take second one.

**Additionals**
If I open camera from some other library in project and then comeback to react-native-camera it starts working fine without any crash. Also it's working fine if some another app is using my microphone.
This error is not related to navigation because actually component is unmounted when I push another screen, also is not related to some feature because I tried removing them all (zoom, flash...) and it's still present.

> Love react-native-camera? Please consider supporting our collective: 👉 https://opencollective.com/react-native-camera/donate
> Want this issue to be resolved faster? Please consider adding a bounty to it https://issuehunt.io/repos/33218414

akai28895, 2022-04-04:
same issue.

ShivaAryal, 2022-06-15:
same issue.