react-native-camera/react-native-camera #3399: Faces Detection is not working in landscape mode, portrait mode it is working fine

thisispranavvasudev, 2021-12-20:
# Bug Report

**To Do First**

- [ ] Did you try latest release? YES
- [ ] Did you try master? YES
- [ ] Did you look for existing matching issues? YES

**Platforms**

 YES
 YES

**Versions**

- Android:
- iOS:
- react-native-camera:
- react-native:
- react:
  

**Description/Current Behaviour**

**Expected Behaviour**

**Steps to Reproduce**

**Additionals**

> Love react-native-camera? Please consider supporting our collective: 👉 https://opencollective.com/react-native-camera/donate
> Want this issue to be resolved faster? Please consider adding a bounty to it https://issuehunt.io/repos/33218414

thisispranavvasudev, 2021-12-31:
Can anybody response to me

UnjustPerson, 2022-06-08:
The issue has been resolved [here](https://github.com/react-native-camera/react-native-camera/issues/2732)

> After some hunting, I found this in RNCameraViewHelper:
```
public static int getCorrectCameraRotation(int rotation, int facing, int cameraOrientation) {
    if (facing == CameraView.FACING_FRONT) {
      // Tested the below line and there's no need to do the mirror calculation
      return (cameraOrientation + rotation) % 360;
    } else {
      final int landscapeFlip = rotationIsLandscape(rotation) ? 180 : 0;
      return (cameraOrientation - rotation + landscapeFlip) % 360;
    }
  }
```

> and thought that maybe it did indeed need the mirror calculation. Changed it to this:

```
public static int getCorrectCameraRotation(int rotation, int facing, int cameraOrientation) {
    if (facing == CameraView.FACING_FRONT) {
      // Tested the below line and there's no need to do the mirror calculation
      final int landscapeFlip = rotationIsLandscape(rotation) ? 180 : 0;
      return (cameraOrientation + rotation + landscapeFlip) % 360;
    } else {
      final int landscapeFlip = rotationIsLandscape(rotation) ? 180 : 0;
      return (cameraOrientation - rotation + landscapeFlip) % 360;
    }
  }
```

thisispranavvasudev, 2022-06-09:
> The issue has been resolved [here](https://github.com/react-native-camera/react-native-camera/issues/2732)
>
>
> ```
> public static int getCorrectCameraRotation(int rotation, int facing, int [...]

Thank you so much @UnjustPerson