react-native-camera/react-native-camera #3402: onBarCodeRead method is getting triggered without scanning QR code

leoamato10, 2021-12-28:
onBarCodeRead method is getting triggered without scanning qr. Its only an android issue.

Im using react-native-qrcode-scanner 1.5.4 and react-native-camera 4.2.1 version. Devices used Samsung s9 and Samsung s21 ultra.

When the RNCamera component is loaded at random time and with the cellphone pointing to the floor it triggers the onBarCodeRead method (I don't know why). It was difficult for me to reproduce this error because I couldn't find a pattern or specific behavior to make it appear.

This is my RNCamera component code:

&lt;<Camera
        androidCameraPermissionOptions={{
          title: this.props.permissionDialogTitle,
          message: this.props.permissionDialogMessage,
          buttonPositive: this.props.buttonPositive,
        }}
        style={[styles.camera, this.props.cameraStyle]}
        onBarCodeRead={this._handleBarCodeRead.bind(this)}
        barCodeTypes={[Camera.Constants.BarCodeType.qr]}
        type={this.props.cameraType}
        flashMode={this.props.flashMode}
        captureAudio={false}
        {...this.props.cameraProps}
      >
        {this._renderCameraMarker()}
      </Camera>
/&gt;

If you have any suggestions or ideas please let me know I will be extremely grateful . Also if you need more information just let me know.

leoamato10, 2021-12-29:
Today I achieved to record the issue and here is the video...

https://user-images.githubusercontent.com/55066576/147677967-fb1e7ba2-53fa-45f4-af77-3fd916dc74bb.MP4

leoamato10, 2021-12-29:
Another Screenshot with error printed on screen:

[image]

CodingWithNobody, 2023-04-15:
Try this https://www.youtube.com/watch?v=sE0MEyFA7E4&list=PLQhQEGkwKZUrempLnmxjt7ZCZJu1W3p2i&index=8