react-native-camera/react-native-camera #3412: Error running android

joaolnpr, 2022-02-09:
Error:

`FAILURE: Build failed with an exception.

* What went wrong:
Could not determine the dependencies of task ':app:compileDebugJavaWithJavac'.
> Could not resolve all task dependencies for configuration ':app:debugCompileClasspath'.
   > Could not resolve project :react-native-camera.
     Required by:
         project :app
      > No matching configuration of project :react-native-camera was found. The consumer was configured to find an API of a component, as well as attribute 'com.android.build.api.attributes.BuildTypeAttr' with value 'debug', attribute 'react-native-camera' with value 'general' but:
          - None of the consumable configurations have attributes.

* Try:
Run with --stacktrace option to get the stack trace. Run with --info or --debug option to get more log output. Run with --scan to get full insights.

* Get more help at https://help.gradle.org

BUILD FAILED in 50s

error Failed to install the app. Make sure you have the Android development environment set up: https://reactnative.dev/docs/environment-setup.
Error: Command failed: gradlew.bat app:installDebug -PreactNativeDevServerPort=8081

error Command failed with exit code 1.
info Visit https://yarnpkg.com/en/docs/cli/run for documentation about this command.`

joaolnpr, 2022-02-09:
**android/build.gradle have** 
   classpath("com.android.tools.build:gradle:4.1.0")

**android/app/build.gradle** 
missingDimensionStrategy 'react-native-camera', 'general'

joaolnpr, 2022-02-09:
Not building when running **yarn android**

joaolnpr, 2022-02-09:
All steps have been done from here: **https://react-native-camera.github.io/react-native-camera/docs/installation.html**

bestinepayyappilly, 2023-02-17:
This is happening when my project was upgraded from 0.67.4 to 0.68.4.
Any updates on why this is happening?

kklobe, 2023-02-27:
I was able to get past this on 0.71.3 by adding a `missingDimensionStrategy` to `android/app/build.gradle`:

```
...
android {
    ndkVersion rootProject.ext.ndkVersion

    compileSdkVersion rootProject.ext.compileSdkVersion

    namespace "com.myapp"
    defaultConfig {
        ...
        // react-native-camera:
        missingDimensionStrategy 'react-native-camera', 'general'
```

CodingWithNobody, 2023-04-15:
Check this https://www.youtube.com/watch?v=8QW5HDa4lQU&list=PLQhQEGkwKZUrempLnmxjt7ZCZJu1W3p2i&index=67