react-native-camera/react-native-camera #3423: invariant violation: viewproptypes has been removed from react native. migrate to viewproptypes exported from 'deprecated-react-native-prop-types'.

jainmca21, 2022-07-14:
Getting below error after upgrading the RN version 0.69.1 when run 'npx react-native run-ios'.

invariant violation: ViewPropTypes has been removed from react native. migrate to ViewPropTypes exported from 'deprecated-react-native-prop-types'.

In React-native version 0.69 'ViewPropTypes'. has been removed.

Essveeyel, 2022-07-15:
Not sure if anyone is maintaining this anymore as stated on the README.

However, if you want to get it working you need to do the fpollowing:
 
 - run `npm i  deprecated-react-native-prop-types ` 
 - go to /node-modules/react-native-camera/src/RNCamera.js
 - delete the imported `ViewPropTypes` from 'react-native'
 - add `import {ViewPropTypes} from 'deprecated-react-native-prop-types'`
 
 This should get it working temporarily.

Aarvis, 2022-07-17:
> Not sure if anyone is maintaining this anymore as stated on the README.
>
> However, if you want to get it working you need to do the fpollowing:
>
> * run `npm i  deprecated-react-native-prop-types `
> * [...]

> Not sure if anyone is maintaining this anymore as stated on the README.
>
> However, if you want to get it working you need to do the fpollowing:
>
> * run `npm i  deprecated-react-native-prop-types `
> * [...]

Worked for me

Almas-Asghar-730, 2022-09-16:
not using react native camera but getting the same error, is their any way to know which module is having the ViewPropTypes

Essveeyel, 2022-09-16:
> not using react native camera but getting the same error, is their any way to know which module is having the ViewPropTypes

Are you not getting any errors in Simulator? The error should point exactly to the file causing the issue.

jkester1986, 2022-09-30:
> not using react native camera but getting the same error, is their any way to know which module is having the ViewPropTypes

@Almas-Asghar-730 another known one is react-native-snap-carousel

JDFranz, 2022-10-08:
Temporary solution works.
How can we make it work by switching to another library to have a long term solution?

Almas-Asghar-730, 2022-10-10:
> > not using react native camera but getting the same error, is their any way to know which module is having the ViewPropTypes
> 
> Are you not getting any errors in Simulator? The error should point exactly to the file causing the issue.

no

Almas-Asghar-730, 2022-10-10:
> > not using react native camera but getting the same error, is their any way to know which module is having the ViewPropTypes
> 
> @Almas-Asghar-730 another known one is react-native-snap-carousel

not using this too

exerror8, 2022-10-19:
[[Solved] ERROR Invariant Violation: ViewPropTypes has been removed from React Native. Migrate to ViewPropTypes exported from ‘deprecated-react-native-prop-types’
](https://exerror.com/error-invariant-violation-viewproptypes-has-been-removed-from-react-native-migrate-to-viewproptypes-exported-from-deprecated-react-native-prop-types/)

jslok, 2022-10-27:
> Temporary solution works. How can we make it work by switching to another library to have a long term solution?

Use the pr instead or branch your own with the fix included.
https://github.com/react-native-camera/react-native-camera/pull/3443

jkester1986, 2022-10-27:
> > Temporary solution works. How can we make it work by switching to another library to have a long term solution?
> 
> Use the pr instead or branch your own with the fix included. #3443 `yarn add react-native-camera/react-native-camera#3443/head` is a drop in replacement

Can also use patch-package

MdImranNazeer1011, 2022-11-14:
i am not using viewproptypes  anywhere in my code but still geting error - **invariant violation: viewproptypes has been removed from react native.**  . and this error occured after install snap carosuel

jkester1986, 2022-11-14:
> I am getting this warning **[ViewPropTypes will be removed from React Native. Migrate to ViewPropTypes exported from 'deprecated-react-native-prop-types](https://stackoverflow.com/questions/71702392/viewproptypes-will-be-removed-from-react-native-migrate-to-viewproptypes-export)**, even I haven't used ViewPropTypes anywhere in my code. and error occured after using snap carousel.

@MdImranNazeer1011 as mentioned already, this is a known issue with that package. You will need to patch it, or fork it, or similar. IIRC there's an open PR in the package with a fix that's been requested for merge for months, it has the solution you need

samih-sghier, 2022-11-22:
@jkester1986 patch-package not working

chandannath98, 2023-01-11:
i am getting the same error 
this is my package.json file

{
  "main": "node_modules/expo/AppEntry.js",
  "scripts": {
    "start": "expo start",
    "android": "expo start --android",
    "ios": "expo start --ios",
    "web": "expo start --web",
    "eject": "expo eject"
  },
  "dependencies": {
    "@expo/vector-icons": "^13.0.0",
    "@react-native-async-storage/async-storage": "~1.17.3",
    "@react-native-community/masked-view": "0.1.10",
    "@react-navigation/bottom-tabs": "^5.11.2",
    "@react-navigation/native": "^5.8.10",
    "@react-navigation/stack": "^5.12.8",
    "expo": "^47.0.0",
    "expo-font": "~11.0.1",
    "expo-linking": "~3.3.0",
    "expo-status-bar": "~1.4.2",
    "moment": "^2.29.1",
    "react": "18.1.0",
    "react-dom": "18.1.0",
    "react-native": "0.70.5",
    "react-native-elements": "^1.2.7",
    "react-native-gesture-handler": "~2.8.0",
    "react-native-paper": "^3.12.0",
    "react-native-reanimated": "~2.12.0",
    "react-native-safe-area-context": "4.4.1",
    "react-native-screens": "~3.18.0",
    "react-native-vector-icons": "^7.1.0",
    "react-native-web": "~0.18.9",
    "react-navigation": "^4.4.3",
    "react-navigation-drawer": "^2.6.0",
    "react-navigation-material-bottom-tabs": "^2.3.3",
    "react-navigation-tabs": "^2.10.1",
    "react-redux": "^7.2.2",
    "redux": "^4.0.4",
    "redux-thunk": "^2.3.0"
  },
  "devDependencies": {
    "babel-preset-expo": "^8.3.0"
  },
  "private": true
}

russelRajitha, 2023-01-18:
why are they really doing this kind of change? i am using tons of libraries and suddenly getting this. is there one fix to apply all of these libraries?

vmsilva, 2023-01-18:
This issue occur because React-native 0.70 > deprecade  ViewPropTypes from react-native, To solve this trouble remove ViewPropTypes import from react-native and them from deprecated-react-native-prop-types, for now this is what can we do.

chandannath98, 2023-01-18:
> i am getting the same error this is my package.json file
>
> { "main": "node_modules/expo/AppEntry.js", "scripts": { "start": "expo start", "android": "expo start --android", "ios": "expo start --ios", [...]

the issue is in my "react-native-elements" package version, after updating the package my issue was solved.

mikeRChambers610, 2023-02-15:
They never fail to make our lives difficult.

anasalaqeel, 2023-02-15:
none of the above works for me because I don't even have react-native-camera module in my app, so I finally fixed it. see the link for the solution
https://stackoverflow.com/a/75463206/12861001

jkester1986, 2023-02-15:
> none of the above works for me because I don't even have react-native-camera module in my app it was from `react-native-snap-carousel` module, so I finally fixed it. see the link for the solution https://stackoverflow.com/a/75463206/12861001

yep, `patch-package` (which is in the stackoverflow solution) has been mentioned, I think people are overlooking it though. `patch-package` is probably the best solution, short of forking whichever repo is causing the problem in order to fix it

KiranSayone, 2023-04-25:
> yep, `patch-package` (which is in the stackoverflow solution) has been mentioned, I think people are overlooking it though. `patch-package` is probably the best solution, short of forking whichever [...]

But don't we need to apply the patch each time we clone the project repo or when we delete the node modules and install the packages using npm install again.