react-native-camera/react-native-camera #3427: Remov

kris1803, 2022-08-08:
Hi! 👋 
      
Firstly, thanks for your work on this project! 🙂

Today I used [patch-package](https://github.com/ds300/patch-package) to patch `react-native-camera@4.2.1` for the project I'm working on.

It talks about deprecated ViewPropTypes in 0.69 React native, they have even been removed from React Native so the app cannot start.

Here is the diff that solved my problem:

```diff
diff --git a/node_modules/react-native-camera/src/RNCamera.js b/node_modules/react-native-camera/src/RNCamera.js
index b7a271a..6dd3123 100644
--- a/node_modules/react-native-camera/src/RNCamera.js
+++ b/node_modules/react-native-camera/src/RNCamera.js
@@ -1,11 +1,11 @@
 // @flow
 import React from 'react';
 import PropTypes from 'prop-types';
+import { ViewPropTypes } from 'deprecated-react-native-prop-types';
 import {
   findNodeHandle,
   Platform,
   NativeModules,
-  ViewPropTypes,
   requireNativeComponent,
   View,
   ActivityIndicator,
```

<em>This issue body was [partially generated by patch-package](https://github.com/ds300/patch-package/issues/296).</em>

DanielGBullido, 2022-08-15:
please include this in next version!

> Hi! 👋
> 
> Firstly, thanks for your work on this project! 🙂
> 
> Today I used [patch-package](https://github.com/ds300/patch-package) to patch `react-native-camera@4.2.1` for the project I'm working on.
> 
> It talks about deprecated ViewPropTypes in 0.69 React native, they have even been removed from React Native so the app cannot start.
> 
> Here is the diff that solved my problem:
> 
> ```diff
> diff --git a/node_modules/react-native-camera/src/RNCamera.js b/node_modules/react-native-camera/src/RNCamera.js
> index b7a271a..6dd3123 100644
> --- a/node_modules/react-native-camera/src/RNCamera.js
> +++ b/node_modules/react-native-camera/src/RNCamera.js
> @@ -1,11 +1,11 @@
>  // @flow
>  import React from 'react';
>  import PropTypes from 'prop-types';
> +import { ViewPropTypes } from 'deprecated-react-native-prop-types';
>  import {
>    findNodeHandle,
>    Platform,
>    NativeModules,
> -  ViewPropTypes,
>    requireNativeComponent,
>    View,
>    ActivityIndicator,
> ```
> 
> _This issue body was [partially generated by patch-package](https://github.com/ds300/patch-package/issues/296)._

wojtekmaj, 2022-08-29:
Clearly not. https://github.com/react-native-camera/react-native-camera#react-native-camera-is-deprecated

azizmobarak, 2022-10-16:
opened a pull request https://github.com/react-native-camera/react-native-camera/pull/3442

bearsworth, 2022-10-20:
I have followed and applied the fix but now I get a unhandled promise rejection now...did this occur or what RN version are you using?

azizmobarak, 2022-10-22:
@bearsworth if you are using RN >67 then go to build.gradle under app folder you will find a line 
`missingDimensionStrategy 'react-native-camera', 'general'` change general to `mlkit`