react-native-camera/react-native-camera #3450: iPhone 14 Face Detection not Working

anveshbabu, 2022-11-07:
in Iphone 14 Pro get face list empty 

err: {type:"face",faces:[],target:99}

getting like this can plz help any one

biladen796, 2022-12-07:
Any one know about this issue? I also have the same issue

mocmoc05, 2023-03-01:
@anveshbabu I spent 1 day to resolve it, I resolved it by following: https://github.com/react-native-camera/react-native-camera/issues/1870#issuecomment-662305158
I think that will save your day too :) @biladen796 @klabusta

fellipe-ribeiro, 2023-03-20:
**> @anveshbabu I spent 1 day to resolve it, I resolved it by following: [#1870 (comment)](https://github.com/react-native-camera/react-native-camera/issues/1870#issuecomment-662305158) I think that will save your day too :) @biladen796 @klabusta**

This change has been tested from iphone 7 to 14 and does it work?

mocmoc05, 2023-03-20:
> **> @anveshbabu I spent 1 day to resolve it, I resolved it by following: [#1870 (comment)](https://github.com/react-native-camera/react-native-camera/issues/1870#issuecomment-662305158) I think that [...]

Yes. it worked

fellipe-ribeiro, 2023-03-20:
> Yes. it worked

Thanks