react-native-camera/react-native-camera #3451: Cannot run android with react-native-camera

nvchien2811, 2022-11-13:
I can't run project when install this lib.
I got error: 
* What went wrong:
Execution failed for task ':react-native-camera:generateGeneralDebugRFile'.
> Could not resolve all files for configuration ':react-native-camera:generalDebugCompileClasspath'.
   > Failed to transform react-native-0.71.0-rc.0-debug.aar (com.facebook.react:react-native:0.71.0-rc.0) to match attributes {artifactType=android-symbol-with-package-name, com.android.build.api.attributes.BuildTypeAttr=debug, org.gradle.category=library, org.gradle.dependency.bundling=external, org.gradle.libraryelements=aar, org.gradle.status=release, org.gradle.usage=java-api}.
      > Execution failed for JetifyTransform: /Users/chien/.gradle/caches/modules-2/files-2.1/com.facebook.react/react-native/0.71.0-rc.0/7a7f5a0af6ebd8eb94f7e5f7495e9d9684b4f543/react-native-0.71.0-rc.0-debug.aar.
         > Java heap space

I tried install by docs:
 missingDimensionStrategy 'react-native-camera', 'general' <-- add this line in android/app/build.gradle
Some time ago I was still able to run my project normally. But since yesterday, I can't run anymore
Can anyone help me!

closed 2022-11-13 by nvchien2811

ahadali, 2023-01-28:
@nvchien2811 did you find any solution for this issue?

nvchien2811, 2023-01-28:
@ahadali I solved it. This error due to  "Java heap space". You can fix it through 
https://stackoverflow.com/questions/69238461/react-native-java-heap-space