react-native-camera/react-native-camera #3480: react-native-camera not running with updated react-native version

Malik-Usman17, 2023-05-02:
So I am building a new react-native app, and installed multiple packages running the app successfully, but when I installed the **react-native-camera** it gives me and error.

`* What went wrong:
Could not determine the dependencies of task ':app:compileDebugJavaWithJavac'.
> Could not resolve all task dependencies for configuration ':app:debugCompileClasspath'.
   > Could not resolve project :react-native-camera.
     Required by:
         project :app
      > The consumer was configured to find an API of a component, preferably optimized for Android, as well as attribute 'com.android.build.api.attributes.BuildTypeAttr' with value 'debug', attribute 'com.android.build.api.attributes.AgpVersionAttr' with value '7.3.1'. However we cannot choose between the following variants of project :react-native-camera:
          - generalDebugApiElements
          - mlkitDebugApiElements
        All of them match the consumer attributes:
          - Variant 'generalDebugApiElements' capability PMIWoodInventory:react-native-camera:unspecified declares an API of a component, as well as attribute 'com.android.build.api.attributes.AgpVersionAttr' with value '7.3.1', attribute 'com.android.build.api.attributes.BuildTypeAttr' with value 'debug':
              - Unmatched attributes:
                  - Provides attribute 'com.android.build.api.attributes.ProductFlavor:react-native-camera' with value 'general' but the consumer didn't ask for it
                  - Provides attribute 'com.android.build.gradle.internal.attributes.VariantAttr' with value 'generalDebug' but the consumer didn't ask for it
                  - Doesn't say anything about its target Java environment (preferred optimized for Android)
                  - Provides attribute 'react-native-camera' with value 'general' but the consumer didn't ask for it
          - Variant 'mlkitDebugApiElements' capability PMIWoodInventory:react-native-camera:unspecified declares an API of a component, as well as attribute 'com.android.build.api.attributes.AgpVersionAttr' with value '7.3.1', attribute 'com.android.build.api.attributes.BuildTypeAttr' with value 'debug':
              - Unmatched attributes:
                  - Provides attribute 'com.android.build.api.attributes.ProductFlavor:react-native-camera' with value 'mlkit' but the consumer didn't ask for it
                  - Provides attribute 'com.android.build.gradle.internal.attributes.VariantAttr' with value 'mlkitDebug' but the consumer didn't ask for it
                  - Doesn't say anything about its target Java environment (preferred optimized for Android)
                  - Provides attribute 'react-native-camera' with value 'mlkit' but the consumer didn't ask for it`
                  
My App Packages:
**react: 18.2.0
react-native: 0.71.7
react-native-camera: ^4.2.1**

uplamartya, 2023-05-17:
Any update on this? facing similar issue

  "react": "18.2.0",
  "react-native": "0.71.8",
  "react-native-camera": "^4.2.1"

Malik-Usman17, 2023-05-21:
> Any update on this? facing similar issue
> 
> "react": "18.2.0", "react-native": "0.71.8", "react-native-camera": "^4.2.1"

Yes I found the solution, you need to add this line in **defaultConfig** block in **android/app/build.gradle** file
**missingDimensionStrategy 'react-native-camera', 'general'**